felixge/node-ar-drone #116: Move specific distance

tlear, 2015-04-08:
Hi,

This may be a stupid question but I can't quite find the answer - how would you suggest going about using ardrone to move the drone a specific amount in a windless environment (say, 1m forward)?  I assume it would be: front(speed) and then stop after a certain time, but I don't see an easy way to determine what that certain time is without knowing the actual speed of the drone, which is not provided.

felixge, 2015-04-08:
This is a hard problem. The drone does not know it's actual position in space, and dead reckoning (what you described) will always have a large amount of error. Your best bet is utilizing the cameras on the drone, but this in itself a challenging task, and I'm not aware of anybody in the nodecopter community having implemented such a system successfully.

shellhead, 2015-04-08:
There is the [ardrone-autonomy](https://github.com/eschnou/ardrone-autonomy) module. You can provide distances to move in a given direct according to the API. I haven't used it, so I can't speak to how accurate it is, but hopefully it's helpful.

felixge, 2015-04-08:
@shellhead ah, right. Completely forgot to mention that. AFAIK it uses dead reckoning, but it's doing that as good as possible :)

wiseman, 2015-04-08:
### General strategy

You need to

1. Start monitoring navdata and establish your termination condition; something like "distance traveled is >= 2000 mm"
2. Start the drone moving.
3. Watch incoming navdata, checking whether your termination condition has been met.
4. Stop the drone when your termination condition has been met.

### Without GPS

Even without GPS, the AR.Drone can make a pretty good guess of its relative position if you're talking about smallish distances (a few meters?).  An [AR.Drone engineer described](https://projects.ardrone.org/boards/1/topics/show/1302) the `navdata.demo.drone.camera.translation` data as  "This position is estimated using all on-board sensors and should be a bit more accurate than integrating the speed information".  I believe they use optical flow from the downward facing camera combined with accelerometer data to come up with a combined best guess position.  This is the data I used to implement commands like "go forward six feet" in [this demo](https://www.youtube.com/watch?v=uhBa11gdbeU).

My [ar-drone-rest library](https://github.com/wiseman/ar-drone-rest) (sorry @andrew, I hadn't seen yours) uses this to offer the following sort of API:

Tell the vehicle to move forward slowly and stop when it has traveled 1 meter:

```
$ curl 'http://127.0.0.1/v1/stopAfter/1000'
$ curl 'http://127.0.0.1/v1/front/0.1'
```

You can find the implemention around https://github.com/wiseman/ar-drone-rest/blob/master/lib/server.js#L37

### With GPS

If you have the [AR.Drone Flight Recorder](http://ardrone2.parrot.com/apps/flight-recorder/) or other GPS, you can measure long distances, though maybe not as precisely.  You would use `navdata.gps.latFuse` and `navdata.gps.lonFuse` to get the coordinates of the drone, and something like [geolib](https://www.npmjs.com/package/geolib) to compute the distance between coordinates.

wiseman, 2015-05-25:
Please re-open if you have more questions on this topic.

closed 2015-05-25 by wiseman