felixge/node-ar-drone #118: Error from PngEncoder: Ffmpeg was not found. Have you installed the ffmpeg package?

nax47, 2015-07-16:
I am using a Windows 8.1 x64 machine and I have ffmpeg installed under "C:\Program Files\ffmpeg". The directory is also added in the path and I don't understand what else I need to do. Does ffmpeg need to be installed in any specific directory?? I need to get a PngStream from my AR Drone 2.0 and the ar-drone node module seems to be working fine otherwise. If there is anything else I need to do please let me know as I don't have much experience working with node.

wiseman, 2015-07-20:
Just to confirm, if you open a command window and try to run ffmpeg from the command line, it works?

nax47, 2015-07-20:
Yes but the .bat file I have set up in my directory is named ffmpeg64, so if I run ffmpeg64 from the command line, yes it works. In any case, I found a workaround by running my node script from ffmpeg itself instead of the node command line. Thanks anyway.

closed 2015-07-20 by nax47