felixge/node-ar-drone #119: only serves one image after every start-up, and "TcpVideoStream Timeout" error will occur.

BigRabbit, 2015-07-20:
1.ffmpeg version 
[image]
2.TcpVideoStream error
[image]
3.run "node test/run.js"
[image]

wiseman, 2015-07-20:
Hi, @BiGrabbit, thanks for the report.  The test failure issue has been reported: See #78 and #49.

The `TcpVideoStream` timeout happens sometimes, but the code is designed to reconnect when that happens, and it looks like that's working as designed in your case.

I'm going to close this for now, but let me know if you don't think I've addressed your issues.

(In the future, please paste text instead of screenshot images--images are not searchable, but text is.)

closed 2015-07-20 by wiseman

BigRabbit, 2015-07-21:
hello, wiseman! thanks for your reply in time. I want to know how to solve the problem "pngSteram returns only one png after every start-up".

wiseman, 2015-07-21:
Ok, for that issue see #99.