felixge/node-ar-drone #121: Send specific altitude to the drone

sulema, 2015-08-07:
Hi,

I am new in this, and I want to give a specific altitude to the drone, I read the libraries that manage the data of navigation but I can not find the answer yet. I think that I need to send data to the ultrasonic but, what do you suggest to me?? 

Thank you.

eschnou, 2015-08-07:
You can't do that with node-ar-drone alone. The only thing you can do is to provide a speed going up or down (just like someone pushing the throttle stick up or down). In order to maintain the altitude you need to add a control layer that observers the altitude navdata and perform adequate control (usually using a PID).

Have a look at https://github.com/eschnou/ardrone-autonomy, which do just that, and more, adding a simple control layer above node-ar-drone.

closed 2015-08-07 by eschnou

wiseman, 2015-08-08:
Also see my answer at https://github.com/felixge/node-ar-drone/issues/116#issuecomment-91006554 for a crude solution.