felixge/node-ar-drone #124: TcpVideoStream error: TcpVideoStream timeout after 1000 ms.

mouhyemen, 2015-08-29:
Despite the patches of #23, I am still facing this trouble. I have the latest node and downloaded the updated library onto my system. Any ideas why the client is not reconnecting and taking care of this?

mouhyemen, 2015-08-30:
So I have made a go at it again. I had node v0.10.33 earlier and now I installed v0.12.7. The client re-connects if it has lost connection and streams in video but the parsing has stopped. In case anyone else faces this problem, try installing to the latest node.

I have checked if TcpVideoStream emits data events after re-connecting and it does. However, the parser has stopped emitting data events.

Has this issue been addressed where once the client loses connection, it is able to reconnect and stream in video from the drone BUT NOT parse it anymore to a h264 file ??

wiseman, 2015-08-30:
I think this is a duplicate of #105.

mouhyemen, 2015-08-31:
Yes it is. I had opened another issue before commenting on #105. Should I close this ?

wiseman, 2015-09-02:
Yes, I'll close this and we'll continue the discussion on #105.  Thanks!

closed 2015-09-02 by wiseman