felixge/node-ar-drone #127: multiple network adapters/interfaces

bno1, 2015-09-03:
My pc has 2 network interfaces, one being the onboard network card connected to the internet and the other begin the wireless adapter that is connected to the drone.

I want all my applications to connect through the onboard adapter and my drone client to connect through the wireless adapter, but the library does not support this very well.

I managed to do this easily for the control and navigation connections by creating 2 custom UDP sockets and creating a new UdpControl and UdpNavdataStream using them.
But for the video stream it's not that easy. The only way to bind a TCP socket to a local address is through net.connect/net.createConnection which doesn't play nice with the way TcpVideoStream uses the socket.

wiseman, 2015-09-04:
Just so I understand, are both network interfaces trying to route the drone's 192.168.1.1 default address?  Because in general it shouldn't be the application deciding which network interface to use, it should be the kernel's network routing.  If you can keep the networks separate, everything should "just work".  For example, my home network is 192.168.100.x, and when I connect my ethernet interface to that and connect my wifi adapter to the drone's network, everything works as you described: Browsers, etc. are able to connect to the internet while the drone client talks to the drone.

Or have I misunderstood what the issue is that you're running into?

bno1, 2015-09-04:
I found the problem. Because I don't haven an wifi adapter I used android's tethering to connect my pc to the drone. Problem is that android sets the network interface ip to 192.168.42.x, so the operating system doesn't know that 192.168.1.1 is actually on the 192.168.42.x network.

bno1, 2015-09-04:
I solved it by running
```
route add 192.168.1.0 mask 255.255.255.0 192.168.42.129
```
This routes all the traffic send to network 192.168.1.x through android's gateway

Thanks for the insights @wiseman

wiseman, 2015-09-04:
Excellent, glad you were able to figure out the solution.

closed 2015-09-04 by wiseman