felixge/node-ar-drone #128: pngStream() low resolution

albertJ32, 2015-10-24:
Hi, 
i'm using pngStream() in a project to analyse textures but i noticed it outputs to 640x360 instead of the normal 1024x720 when taking stills with the ar drone app. Is this normal and is there a solution?

Thanks

darsnack, 2016-10-25:
I have noticed a similar situation. I did some testing, and I found out the 640x360 is because the drone is using the bottom camera instead of the front facing camera. I tried setting the front facing camera in the config using client.config('video:video_channel', 0), but I can verify that it is still using the bottom camera.

CianDuffy, 2017-02-20:
I'm also using the front facing camera and the resolution is still 640x360.

Kiwicasper, 2017-04-05:
You can set the image quality by passing an option when creating the client. Like:
var option = new Object();
option.imageSize = "1280x720";
var client = arDrone.createClient(options);