felixge/node-ar-drone #131: What's going on?

giggioz, 2016-05-09:
Can i aks to the developers why this repo is not updated since a long time?

It seems a very good  project.

Is there a better alternative around here?

Thanks.

dwiyatci, 2016-05-09:
Second this! It's been almost two years ago since the last commit 😶

wiseman, 2016-05-09:
I've been wondering/thinking about this myself lately.  http://dronegames.co/ hasn't even been updated since, like, 2014.  Are people not into small programmable drones anymore?  Have they moved on the the Bebop or [other Parrot drones](https://github.com/voodootikigod/node-rolling-spider), or vehicles by other manufacturers?  Or maybe this library has just gotten to the point where it does enough of what people want it to do that more development isn't really necessary.

I personally am still interested in programming the AR.Drone, though I've spent more of my time lately on a [clojure port](https://github.com/wiseman/turboshrimp) of node-ar-drone.  If you have some ideas for improvements for node-ar-drone, I'd be willing to take a look at it or at least help you work on it.

felixge, 2016-05-12:
I'm the original developer of this library. I can't speak for other contributors, but my interests have shifted. I'm now developing in Go instead of node.js, and my recent side-projects have been more about 3d printing and electronics.

Dronegames piggy backed on the NodeCopter idea, but threw our original core values out of the window. As far as I can tell it was abandoned after its creator turned the resulting attention in VC funding for his own drone startup. This experience has also led to some disillusion about writing open source software and organizing community events for me.

That being said, I'm going to mentor a young high school student for a 2 month drone / computer vision project this summer. So perhaps I'll end up making some changes to this library again.

giggioz, 2016-05-12:
Ehy felixge, thank you for your follow up, i appreciate.

It's a shame hearing such stories but the reality shows that open source is driving the world to a better place so... don't be discouraged :)

It's nice to see that someone after you is going to keep this project up!

Regards!

wiseman, 2016-07-13:
Closing this since I think the question has been adequately answered 😄

closed 2016-07-13 by wiseman