felixge/node-ar-drone #133: Supported drones 2016

clankill3r, 2016-05-13:
I was wondering which drones are supported besides the Parrot AR drone 2.0?
Please list here if you know a drone with support.

And if the parrot bebop 2 for example is not supported. Then how can be checked if it is possible to program support for it?

felixge, 2016-05-13:
AR Drone 2 and 1 are supported

closed 2016-05-13 by felixge

tanmayb123, 2016-09-23:
Quick Question - Is any edition of the AR Drone 2 supported? I can get access to an AR Drone Power Edition tomorrow (http://www.bestbuy.ca/en-CA/product/parrot-parrot-ar-drone-2-0-power-edition-quadcopter-drone-with-camera-ready-to-fly-red-pf721005bi/10273736.aspx?lang=en-CA&pcname=&sku=10273736&path=3153be675a3f25aecb458530d5f0d034en02), but not an Elite Edition. Would that work? Just wanted to make sure before going ahead and buying it...

Thanks!

felixge, 2016-09-23:
There is no technical difference between the editions. The only thing different is color and included batteries.

tanmayb123, 2016-09-23:
Alright - Thank you very much!