felixge/node-ar-drone #138: client.config('video:video_channel', 0) not setting front facing camera

darsnack, 2016-10-25:
I had some code that used the bottom camera, and it worked fine. I changed the code to use the front camera using
client.config('video:video_channel', 0);
but this does not seem to work.

For reference, my drone is doing object tracking. When I use the front facing camera code above, the drone is unresponsive to the object moving in front of it. However, if I keep the code the same, and I place the object under the drone, it reacts. Any ideas why the above command could be failing? I even placed my code in an async.waterfall block to make sure the camera was configured first before I had the drone takeoff and enter the main control loop.

paulhayes, 2017-10-07:
I'm getting the same issue. Any progress on it?

felixge, 2017-10-10:
Yeah, it's possible the FW changed. Parrot used to release a detailed PDF along with their SDK, maybe somebody can hunt down the latest version and see what it says.