felixge/node-ar-drone #141: Is it possible to insert return home gps data in ar drone flight recorder?

JinHan1030, 2017-02-03:
Hello I am a student studying ar drone parrot.
Using the appropriate code, I obtained the GPS data from the parrot ar drone.
But i want to change the returned GPS data by putting it in a parrot ar drone or flight recorder.
Is it possible...?
Thank  you for reading!

wiseman, 2017-04-10:
You want the drone to fly home to specific GPS coordinates?

YunJinHan, 2017-04-14:
JinHna1030 is me , anyway that's right..
i can success move my drone to specific waypoints using other way ,
But i want to know that it is possible for parrot to fly home to specific GPS coordinate..
If you know that answer , Please answer how to fly parrot drone to specific GPS coordinates using ar drone module ...
Thank you for reading wiseman!

wiseman, 2017-04-14:
There are two ways you could do this.

First, the hard way. The protocol used by `node-ar-drone` to talk to the AR.Drone doesn't have any way to directly give the drone coordinates to fly to, but you could do the calculations yourself.

1. Enable the drone's magnetometer (see https://github.com/wiseman/webflight-traffic/blob/master/index.js#L19).
2. Calibrate the magnetometer: Command the drone to take off, then do `client.calibrate(0)` (see https://github.com/felixge/node-ar-drone#clientcalibratedevice_num).
3. Get the drone's magnetic heading from `navdata.magneto.heading.fusionUnwrapped`.
4. Get the drone's current coordinates from `navdata.gps.latitude` and `navdata.gps.longitude`.
5. Compute the desired heading to get from the drone's current position to the desination.  There are subtleties to this, but for most applications you can follow the procedure described in http://www.movable-type.co.uk/scripts/latlong.html:

    > θ = atan2( sin Δλ ⋅ cos φ2 , cos φ1 ⋅ sin φ2 − sin φ1 ⋅ cos φ2 ⋅ cos Δλ ) where φ1,λ1 is the starting point, and φ2,λ2 is the end point (Δλ is the difference in longitude).
    >
    > Since atan2 returns values in the range -π ... +π (that is, -180° ... +180°), to normalise the result to a compass bearing (in the range 0° ... 360°, with −ve values transformed into the range 180° ...
 360°), convert to degrees and then use (θ+360) % 360, where % is (floating point) modulo.

6. Figure out if the drone needs to rotate clockwise or counter-clockwise to achieve the desired heading, and call `client.clockwise` or `client.counterClockwise` as appropriate.

7. When the drone is facing the correct heading, tell it to move forward with `client.front`.

Once the drone is headed in the right direction, every time you get new navdata, you should check the current heading, recompute the desired heading, and tell the drone to turn clockwise or counter-clockwise if needed.  Once you're within some threshold (5 feet? 50 feet?) of your desired destination, tell the drone to stop with `client.stop`.

Besides the protocol that `node-ar-drone` uses, the AR.Drone also supports the [MAVLink](http://qgroundcontrol.org/mavlink/start) protocol, used by many hobbyist and commercial drones.  That protocol has a way to tell the drone to move directly to specific coordinates.  You could use an application like [QGroundControl](http://qgroundcontrol.com/) to send the drone to a particular latitude, longitude, or you can use one of the existing MAVLink libraries to send messages to the drone to do the same thing.

DroneKit-Python, for example, is a high-level library that lets you write code like this (taken from [this example](http://python.dronekit.io/examples/simple_goto.html)):

```
# set the default travel speed
vehicle.airspeed=3
    
point1 = LocationGlobalRelative(-35.361354, 149.165218, 20)
vehicle.simple_goto(point1)
```

closed 2017-04-14 by wiseman

YunJinHan, 2017-04-14:
Thank you for your helping wisemen!

wiseman, 2017-04-14:
(I don't know if it's you who asked basically the same question on stackoverflow, but I gave the same answer there: http://stackoverflow.com/questions/42897828/how-can-i-send-the-gps-data-to-parrot-ar-drone-and-go-to-my-drone-to-gps-positio/43334387#43334387)