felixge/node-ar-drone #142: Vision Dectect..TypeError: Cannot read property 'nbDetected' of undefined

jbakerII, 2017-03-18:
Hi:

I am trying to learn the vision detect functions with the ar Drone but I am  getting the error: "TypeError: Cannot read property 'nbDetected' of undefined" with the code below:

var arDrone = require('ar-drone'); 
var arDroneConstants = require('ar-drone/lib/constants');

// Connect and configure the drone
var client = new arDrone.createClient();
client.config('control:flying_mode', '1');
client.config('detect:detect_type', '12');
client.config('video:video_channel','1');

console.log("....looking for the roundel...");

//add a handler on the navdata updates
client.on('navdata', function(d) {

	
	
	if (d.visionDetect.nbDetected>0) {
		console.log("Detected: %j", d.visionDetect );
	}
	
})

Please help.

Thanks.

Talamantez, 2017-03-23:
can you console.dir (d) ?

jbakerII, 2017-03-24:
Hi:

I was successful with  "console.dir (d)". See the output below:

droneState:
 { flying: 0,
   videoEnabled: 0,
   visionEnabled: 0,
   controlAlgorithm: 0,
   altitudeControlAlgorithm: 1,
   startButtonState: 0,
   controlCommandAck: 0,
   cameraReady: 1,
   travellingEnabled: 0,
   usbReady: 0,
   navdataDemo: 1,
   navdataBootstrap: 0,
   motorProblem: 0,
   communicationLost: 0,
   softwareFault: 0,
   lowBattery: 0,
   userEmergencyLanding: 0,
   timerElapsed: 0,
   MagnometerNeedsCalibration: 0,
   anglesOutOfRange: 0,
   tooMuchWind: 0,
   ultrasonicSensorDeaf: 0,
   cutoutDetected: 0,
   picVersionNumberOk: 1,
   atCodecThreadOn: 1,
   navdataThreadOn: 1,
   videoThreadOn: 1,
   acquisitionThreadOn: 1,
   controlWatchdogDelay: 0,
   adcWatchdogDelay: 0,
   comWatchdogProblem: 0,
   emergencyLanding: 0 },
sequenceNumber: 1528,
visionFlag: 1,
demo:
 { controlState: 'CTRL_LANDED',
   flyState: 'FLYING_OK',
   batteryPercentage: 75,
   rotation:
    { frontBack: 1.765,
      pitch: 1.765,
      theta: 1.765,
      y: 1.765,
      leftRight: 4.197,
      roll: 4.197,
      phi: 4.197,
      x: 4.197,
      clockwise: 2.136,
      yaw: 2.136,
      psi: 2.136,
      z: 2.136 },
   frontBackDegrees: 1.765,
   leftRightDegrees: 4.197,
   clockwiseDegrees: 2.136,
   altitude: 0,
   altitudeMeters: 0,
   velocity: { x: -0.2444150596857071, y: -0.1564984768629074
   xVelocity: -0.2444150596857071,
   yVelocity: -0.1564984768629074,
   zVelocity: 0,
   frameIndex: 0,
   detection: { camera: [Object], tagIndex: 0 },
   drone: { camera: [Object] } },
visionDetect:
 { nbDetected: 0,
   type: [ 0, 0, 0, 0 ],
   xc: [ 0, 0, 0, 0 ],
   yc: [ 0, 0, 0, 0 ],
   width: [ 0, 0, 0, 0 ],
   height: [ 0, 0, 0, 0 ],
   dist: [ 0, 0, 0, 0 ],
   orientationAngle: [ 0, 0, 0, 0 ],
   rotation: [ [Object], [Object], [Object], [Object] ],
   translation: [ [Object], [Object], [Object], [Object] ],
   cameraSource: [ 0, 0, 0, 0 ] } }
header: 1432778632,

The field "visionDetect.nbDetected" is listed but  I am still getting the same error:
TypeError: Cannot read property 'nbDetected' of undefined
    at Client.<anonymous> (C:\Users\kellogg\Documents\Drone Programming\Navdata.
js:32:49)
    at emitOne (events.js:96:13)
    at Client.emit (events.js:188:7)
    at Client._handleNavdata (C:\Users\kellogg\Documents\Drone Programming\node_
modules\ar-drone\lib\Client.js:120:8)
    at emitOne (events.js:96:13)
    at UdpNavdataStream.emit (events.js:188:7)
    at UdpNavdataStream._handleMessage (C:\Users\kellogg\Documents\Drone Program
ming\node_modules\ar-drone\lib\navdata\UdpNavdataStream.js:73:10)
    at emitTwo (events.js:106:13)
    at Socket.emit (events.js:191:7)
    at UDP.onMessage (dgram.js:550:8)

Please advise.

Thanks

Talamantez, 2017-03-24:
It looks like "visionDetect.nbDetected" is 0.  According to this thread, I think this feature may have been abandoned during development. Good luck!

https://github.com/felixge/node-ar-drone/issues/53

wiseman, 2017-03-29:
If you print out the navdata object and format it nicely, you'll see that the value you want is actually `d.demo.visionDetect.nbDetected`--note the intervening **demo**.

Issue #53 talks about what you need to do to enable vision processing. As far as I know the feature works fine.

jbakerII, 2017-03-30:
Hi John:

Thanks for the tip and response. I dug a bit deeper as I have no doubt that
the functionality works. It seems that my code is not pulling in the full
set of "demo" data when I go to the "demo" level. Please see my code and
output below:

var drone = require("ar-drone")
//var arDroneConstants = require('./node_modules/ar-drone/lib/constants');
var arDroneConstants = require('ar-drone/lib/constants');
//var exiting = false;

var client = drone.createClient();

/*
var options = (
1 << constants.options.DEMO |
1 << constants.options.VISION_DETECT |
1 << constants.options.MAGNETO |
1 << constants.options.WIFI |
1 << constants.options.ALTITUDE |
1 << constants.options.ZIMMO_3000
);

*/

function navdataOptionsMask(c) {

  return 1<<c;
}

var navdata_options= (
  navdataOptionsMask(arDroneConstants.options.DEMO) |
  navdataOptionsMask(arDroneConstants.options.VISION_DETECT) |
  navdataOptionsMask(arDroneConstants.options.MAGNETO) |
  navdataOptionsMask(arDroneConstants.options.WIFI)

);

client.config('general:navdata_demo', true);
//client.config('general:navdata_options', options);

client.config('general:navdata_options', navdata_options);

client.config('video:video_channel', 1);
client.config('detect:detect_type', '12');

//All Nav Data
//client.on('navdata', console.log);

client.on('navdata', function(d) {
  console.log("========================ALL NAV
DATA==============================")
  console.log(d);
  //console.log("Flying:  " + d.droneState.flying);

  if (d.demo) {
    console.log("========================ALL NAV DEMO
DATA==============================")
    console.log(d.demo);
  }

// end on ctrl-c
process.on ('SIGINT', function() {

if (exiting) {

process.exit(0);
} else {
console.log('Ending the routine to read NAV Data from the Drone....bye..');
exiting = true;
};

});
});

Output:

========================ALL NAV DATA==============================
{ header: 1432778632,
  droneState:
   { flying: 0,
     videoEnabled: 0,
     visionEnabled: 0,
     controlAlgorithm: 0,
     altitudeControlAlgorithm: 1,
     startButtonState: 0,
     controlCommandAck: 0,
     cameraReady: 1,
     travellingEnabled: 0,
     usbReady: 0,
     navdataDemo: 1,
     navdataBootstrap: 0,
     motorProblem: 0,
     communicationLost: 0,
     softwareFault: 0,
     lowBattery: 0,
     userEmergencyLanding: 0,
     timerElapsed: 0,
     MagnometerNeedsCalibration: 0,
     anglesOutOfRange: 0,
     tooMuchWind: 0,
     ultrasonicSensorDeaf: 0,
     cutoutDetected: 0,
     picVersionNumberOk: 1,
     atCodecThreadOn: 1,
     navdataThreadOn: 1,
     videoThreadOn: 1,
     acquisitionThreadOn: 1,
     controlWatchdogDelay: 0,
     adcWatchdogDelay: 0,
     comWatchdogProblem: 0,
     emergencyLanding: 0 },
  sequenceNumber: 4014,
  visionFlag: 1,
  demo:
   { controlState: 'CTRL_LANDED',
     flyState: 'FLYING_OK',
     batteryPercentage: 100,
     rotation:
      { frontBack: -2.165,
        pitch: -2.165,
        theta: -2.165,
        y: -2.165,
        leftRight: 1.443,
        roll: 1.443,
        phi: 1.443,
        x: 1.443,
        clockwise: 7.377,
        yaw: 7.377,
        psi: 7.377,
        z: 7.377 },
     frontBackDegrees: -2.165,
     leftRightDegrees: 1.443,
     clockwiseDegrees: 7.377,
     altitude: 0,
     altitudeMeters: 0,
     velocity: { x: -0.27344563603401184, y: 2.7715163230895996, z: 0
     xVelocity: -0.27344563603401184,
     yVelocity: 2.7715163230895996,
     zVelocity: 0,
     frameIndex: 0,
     detection: { camera: [Object], tagIndex: 0 },
     drone: { camera: [Object] } },
  visionDetect:
   { nbDetected: 0,
     type: [ 0, 0, 0, 0 ],
     xc: [ 0, 0, 0, 0 ],
     yc: [ 0, 0, 0, 0 ],
     width: [ 0, 0, 0, 0 ],
     height: [ 0, 0, 0, 0 ],
     dist: [ 0, 0, 0, 0 ],
     orientationAngle: [ 0, 0, 0, 0 ],
     rotation: [ [Object], [Object], [Object], [Object] ],
     translation: [ [Object], [Object], [Object], [Object] ],
     cameraSource: [ 0, 0, 0, 0 ] },
  magneto:
   { mx: -56,
     my: -82,
     mz: 68,
     raw: { x: 286.359375, y: 195.5625, z: -246.234375 },
     rectified: { x: 212.90121459960938, y: 190.27171325683594, z: -24
,
     offset: { x: 73.45816802978516, y: 5.29079008102417, z: 0 },
     heading:
      { unwrapped: 0,
        gyroUnwrapped: 0,
        fusionUnwrapped: 7.377906322479248 },
     ok: 1,
     state: 2,
     radius: 217.13441467285156,
     error: { mean: 71.75249481201172, variance: 79.17041015625 } },
  wifi: { linkQuality: 1 } }
========================ALL NAV DEMO DATA=============================
{ controlState: 'CTRL_LANDED',
  flyState: 'FLYING_OK',
  batteryPercentage: 100,
  rotation:
   { frontBack: -2.165,
     pitch: -2.165,
     theta: -2.165,
     y: -2.165,
     leftRight: 1.443,
     roll: 1.443,
     phi: 1.443,
     x: 1.443,
     clockwise: 7.377,
     yaw: 7.377,
     psi: 7.377,
     z: 7.377 },
  frontBackDegrees: -2.165,
  leftRightDegrees: 1.443,
  clockwiseDegrees: 7.377,
  altitude: 0,
  altitudeMeters: 0,
  velocity: { x: -0.27344563603401184, y: 2.7715163230895996, z: 0 },
  xVelocity: -0.27344563603401184,
  yVelocity: 2.7715163230895996,
  zVelocity: 0,
  frameIndex: 0,
  detection:
   { camera: { rotation: [Object], translation: [Object], type: 12 },
     tagIndex: 0 },
  drone: { camera: { rotation: [Object], translation: [Object] } } }

Please advise.

Thanks.

Jesse Baker

wiseman, 2017-03-30:
Jesse, there's something wrong with the formatting of your output--when I ran it through http://jsbeautifier.org/ it showed that `visionDetect` is inside `demo`, but that's not the case--it's just that your output is missing characters for some reason.  So I think `navdata.visionDetect.nbDetected` is correct.

If that doesn't work for you, about all I can think of is that `visionDetect` is only being sent in the data sometimes.  You might try something like

```
if (navdata.visionDetect) {
  console.log(navdata.visionDetect.nbDetected);
}
```

(As a meta-note, it's not easy to read your source code--it'd be easier for people to help you if you cleaned it up before posting it--get rid of excess empty lines, remove commented out blocks, use indentation and standard formatting, as output by tools like jsbeautifier.org etc.  You might need to reply on github using Markdown instead of via email to get good formatting.)

jbakerII, 2017-03-31:
Hi John:

Thanks for the tips. I found the visiondetect settings on the navdata root and not within demo. I was able to detect successfully.

closed 2017-03-31 by jbakerII

wiseman, 2017-03-31:
Awesome, glad you solved it!