felixge/node-ar-drone #144: How do I access just the gps.latitude property?

khunkovic, 2017-03-29:
when I try this. 

var arDrone = require('ar-drone');
var droneClient = arDrone.createClient();
droneClient.config('general:navdata_demo', 'FALSE'); // get back all data the copter can send
droneClient.config('general:navdata_options', 777060865); // turn on GPS

droneClient.on('navdata', function(navdata) {
   console.log(navdata.gps.latitude + ', ' + navdata.gps.longitude);
   // do stuff with the GPS information....
});
droneClient.takeoff(); .....

I get this error 
C:\data>node gpsLog.js C:\data\gpsLog.js:8 console.log(navdata.gps.latitude + ', ' + navdata.gps.longitude); ^ TypeError: Cannot read property 'latitude' of undefined at Client.<anonymous> (C:\data\gpsLog.js:8:27) at emitOne (events.js:96:13) at Client.emit (events.js:188:7) at Client._handleNavdata (C:\data\node_modules\ar-drone\lib\Client.js:194:8) at emitOne (events.js:96:13) at UdpNavdataStream.emit (events.js:188:7) at UdpNavdataStream._handleMessage (C:\data\node_modules\ar-drone\lib\navdata\UdpNavdataStream.‌​js:73:10)

if I change 1 line of code I get back all gps data. I just want the latitude. 
console.log(navdata.gps);

Talamantez, 2017-03-30:
Hi - if you want the nav data, I think you need to change line 3: 'FALSE'  to 'TRUE'

khunkovic, 2017-03-30:
I'm getting nav data. I'm getting all of it in fact when I use console.log(navdata.gps);
What I want to do is just isolate the latitude data. so that it can be saved to 1 variable and  used in making a decision. 
Fyi. general nav data FALSE makes it return everything. general nav data TRUE will limit the data.
Does anyone know how to only return the navdata.gps.latitude ?
this article says its possible however it does not seem to work this way. http://stackoverflow.com/questions/24170884/get-gps-data-from-ar-drone-2-0

khunkovic, 2017-04-08:
With a small mod adding a try catch block I get error, then data then error. Is this a bug?
the gps data only transmits for a while. then it quits.

var arDrone = require('ar-drone');
var droneClient = arDrone.createClient();
droneClient.config('general:navdata_demo', 'FALSE'); // get back all data the copter can send
droneClient.config('general:navdata_options', 777060865); // turn on GPS

droneClient.on('navdata', function(navdata) {

  try {
    console.log(navdata.gps.latitude + ', ' + navdata.gps.longitude);
}
catch(err) {
    console.log(err.message);
}

});

output. 

Cannot read property 'latitude' of undefined 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 43.6330829, -116.4467156 Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' of undefined Cannot read property 'latitude' 
Does anyone know why this happens?

Talamantez, 2017-04-09:
can you print out navdata & navdata.gps?

khunkovic, 2017-04-10:
Here is navdata.gps. 
C:\data>node gpsLog.js
undefined
undefined
undefined
undefined
{ latitude: 43.6330202,
  longitude: -116.4466826,
  elevation: 786.9,
  hdop: 1,
  dataAvailable: 7,
  zeroValidated: 1,
  wptValidated: 0,
  lat0: 43.6330202,
  lon0: -116.4466826,
  latFuse: 43.6330029,
  lonFuse: -116.446652,
  gpsState: 1,
  xTraj: 0,
  xRef: 0,
  yTraj: 0,
  yRef: 0,
  thetaP: 0,
  phiP: 0,
  thetaI: 0,
  phiI: 0,
  thetaD: 0,
  phiD: 0,
  vdop: 0,
  pdop: 0,
  speed: 0.10000000149011612,
  lastFrameTimestamp: 586.815,
  degree: 76.62000274658203,
  degreeMag: 0,
  ehpe: 3.3499999046325684,
  ehve: 0.3799999952316284,
  c_n0: 34,
  nbSatellites: 8,
  channels:
   [ { sat: 3, cn0: 37 },
     { sat: 16, cn0: 36 },
     { sat: 22, cn0: 32 },
     { sat: 9, cn0: 30 },
     { sat: 4, cn0: 31 },
     { sat: 23, cn0: 32 },
     { sat: 7, cn0: 18 },
     { sat: 26, cn0: 26 },
     { sat: 14, cn0: 0 },
     { sat: 31, cn0: 30 },
     { sat: 6, cn0: 23 },
     { sat: 138, cn0: 0 } ],
  gpsPlugged: 1,
  ephemerisStatus: 0,
  vxTraj: 0,
  vyTraj: 0,
  firmwareStatus: 1 }
{ latitude: 43.6330202,
  longitude: -116.4466826,
  elevation: 786.9,
  hdop: 1,
  dataAvailable: 7,
  zeroValidated: 1,
  wptValidated: 0,
  lat0: 43.6330202,
  lon0: -116.4466826,
  latFuse: 43.6330029,
  lonFuse: -116.446652,
  gpsState: 1,
  xTraj: 0,
  xRef: 0,
[... 1225 lines cut ...]
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined
undefined