felixge/node-ar-drone #147: How to add delay to navdata read, and make real time graph?

ivejun, 2017-04-25:
how to add delay for navdata read? because its too fast.
what can i use if i want build real time graph?

here is my Code.

`var arDrone = require('ar-drone'); 

var client = new arDrone.createClient();
client.on('navdata', function(d){
    if(d.demo){
		console.log('pitch:' + d.demo.rotation.roll);
    }
	});`

YunJinHan, 2017-05-01:
i think if you want to just read slow navdata data , using setInterval / setTimeout function

for example

setInterval(function(){
  droneClient.on('navdata', function(navdata) {
    try {
      console.log('test:' + navdata.demo.velocity.x);
    }
    catch(err) {
      console.log(err.message);
    }
  });
},1000);

or 

setInterval(function(){
  droneClient.on('navdata', function(navdata) {
    try {
      console.log('test:' + navdata.demo.velocity.x);
    }
    catch(err) {
      console.log(err.message);
    }
  });
  setTimeout(function(){
    droneClient.removeListener('navdata',function(){
    });
  },1000);
},1000);

the above code prints navdata data every 1 second. 

may be? it's work

wiseman, 2017-05-02:
Just ignore any navdata you don't want to process.

closed 2017-05-02 by wiseman

ivejun, 2017-05-04:
your suggest about set interval didn't work at all. its still read nav data too fast

wiseman, 2017-05-04:
You can't really change the rate at which navdata comes in.  You can just ignore navdata if you don't want to process it.  The code below only processes navdata once per second:

```
var lastNavDataMs = 0;

client.on('navdata', function(d) {
  var nowMs = new Date().getTime();
  if (nowMs - lastNavDataMs > 1000) {
    lastNavDataMs = nowMs;
    // Process navdata once per second.
    if(d.demo){
      console.log('pitch:' + d.demo.rotation.roll);
    }
  }
});
```