felixge/node-ar-drone #152: Multiple clients - multiple drones communicate through nodejs server

craciunraluca94, 2017-06-08:
Hello,

I want to make a web application which manages AR drones. I want to have multiple clients which can make an account and get data from their drones in browser. Is this posible? Can I identify each drone unique or can i make an array of ardrone clients  ( var client = arDrone.createClient(); ) ?

I need some opinions. Thank you !!

felixge, 2017-06-09:
Hi. In theory you can have an application that talks to multiple drones. That being said, they will all have to be on the same network in order for the application to be able to talk to them. Depending on your use cases for the application, that might be a problem.

You won't be able to use this library on the client side. The error message you got is because Browsers don't allow JS to do UDP networking, which is needed for talking to AR Drones.

craciunraluca94, 2017-06-09:
I see.Thanks. Maybe I'll make something like : clients have the whole app (server(+db) + client) on their PCs ( like local server). Sounds it weird ?